CollectiveOpenSourceHardware/LibreSolarBox #5: integration of solarpanel holder

mikelovic, 2017-09-26:
@case06 @DhavalGambhava 
just a thinking...
extend the batteriebox frame with a holder system for solarpanels. also based on UniProKit.

case06, 2017-09-26:
Do you mean something for clipping panels on the rooftop or something like a standalone-holder ? The latter could be interesting by combining it with a tracking system that moves the panel around, following the sun path. This could give you about 30% more energy harvested plus maybe 10% from the MPPT.

But there is one drawback: on windy/stormy days there are enormous forces directed towards the holder mounting because the panels have a lot of surface area where the wind can hook on.
Difficult to calculate the needed static (beyond my scope, this is engineers-stuff ;))

DhavalGambhava, 2017-09-28:
@mikelovic  I updated older design that can hold solar panel. Have you finalize solar module? If yes please send me name and specifications.

mikelovic, 2017-09-29:
@DhavalGambhava  yeah looks good. would be the easiest way to fix the panel near the box.
@case06 for a tracking system we would need some more parts (stepper, µC, ...). but yeah would be nice for sure. i think at the first step it would just be nice to have the opportunity to fix a panel.

case06, 2017-09-30:
btw. what type of inverter have you forseen ? 

My recommendation for an entry-model is a Victron Phoenix 350/12V with 300W constant power (or the 800/12V version) and true sinus. I have already tested it in a productive environment and other projects like sunzilla uses it also, cause its known for its good quality.

link to datasheet: https://www.victronenergy.com/upload/documents/Datasheet-Phoenix-Inverter-180VA-1200VA-EN.pdf

link to dimensions (for freecad-modeling): https://www.victronenergy.com/upload/documents/Dimensions-Phoenix-Inverter-350.pdf

and/or

https://www.victronenergy.com/upload/documents/Dimensions-Phoenix-Inverter-800-1200.pdf

case06, 2017-09-30:
i have uploaded some pictures that show, how the junction of Dhajvals panelholder can be done in UniProKit.

https://github.com/CollectiveOpenSourceHardware/LibreSolarBox/tree/master/boxframe/panelsupport

and i am working on a upklib-based boxframe , have also uploaded some pics and freecad-files , see

https://github.com/CollectiveOpenSourceHardware/LibreSolarBox/tree/master/boxframe/upkframe

case06, 2017-09-30:
@DhavalGambhava   I have some 100W panels here, which have the outer dimensions 106 x 67 x 3.6 cm

DhavalGambhava, 2017-10-05:
@case06  I designed solar panel holder for 160 x 100 x 4 cm solar panel, and it can be also fit for smaller panel as well.

DhavalGambhava, 2017-12-16:
@mikelovic 
@case06 
We have flexible solar panel, so Holder can't hold perfectly to solar panel. Better to not use solar panel holder.

case06, 2017-12-16:
since the holder is sort of "telescopic" it could be used to give the flexible panel some tension. ;)

mikelovic, 2017-12-19:
i want to keep the solar panel holder as an optional module if someone wants to use not flexible solar panels.