CollectiveOpenSourceHardware/LibreSolarBox #12: Inverter

mikelovic, 2017-09-30:
@case06 wrote:
btw. what type of inverter have you forseen ?

My recommendation for an entry-model is a Victron Phoenix 350/12V with 300W constant power (or the 800/12V version) and true sinus. I have already tested it in a productive environment and other projects like sunzilla uses it also, cause its known for its good quality.

link to datasheet: https://www.victronenergy.com/upload/documents/Datasheet-Phoenix-Inverter-180VA-1200VA-EN.pdf

link to dimensions (for freecad-modeling): https://www.victronenergy.com/upload/documents/Dimensions-Phoenix-Inverter-350.pdf

and/or

https://www.victronenergy.com/upload/documents/Dimensions-Phoenix-Inverter-800-1200.pdf

mikelovic, 2017-09-30:
i found on conrad this one.
https://www.conrad.de/de/netzwechselrichter-victron-energy-phoenix-12250-250-w-12-vdc-92-bis-17-v-schraubklemmen-schutzkontakt-steckdose-kab-1556057.html

250W output power is enough.
@martinjaeger @case06 would this one be good?

case06, 2017-09-30:
yes this would be also ok, but the 350/12V can be cheaper than this ;)

eg. here: https://www.photovoltaik4all.de/victron-phoenix-12/350-12v-dc-/-350va-ac

or here:

https://www.amazon.de/Phoenix-Wechselrichter-Schuko-Steckdose-Spannungswandler/dp/B01DT2I1S8/ref=pd_cp_60_1?_encoding=UTF8&psc=1&refRID=X4CF2RYJNEDA24SCN1B0

For 20% lower price you get 50% more output  ;)

closed 2017-11-28 by mikelovic