CollectiveOpenSourceHardware/LibreSolarBox #14: Visualization

mikelovic, 2017-10-04:
visualization based on @openenergymonitor .
media: smart phone, tablet
communication: mppt -> esp; esp -> wlan-router

ThulasiKrishna, 2017-10-20:
Using Standalone Raspberry Pi for Monitoring.
 Process flow : 
Solar Box --> Raspberry pi (Via Serial connector) ----> Push data to Energymonitor locally.

mikelovic, 2017-12-14:
new concept:
- extra PCB as CAN/Raspberry interface -> see: https://github.com/LibreSolar/CAN_RPi-ZeroW
- data send via CAN. stored on RasbperryPiZero as server
- access on Rasbperry via Wifi
- OEM server installed on Rasberry

mikelovic, 2019-03-20:
standard connnection to oem, solar-pv template, https://guide.openenergymonitor.org/applications/solar-pv/

- [ ] preparing parameter string
- [ ] function modul for easy setup options