CollectiveOpenSourceHardware/LibreSolarBox #15: Box Design

mikelovic, 2017-10-06:
@DhavalGambhava  please take the layout from /batterycase from @case06 and extend it with

- components: MPPT, BMS, inverter, batteries
- plugs: SolarInput, DC-Output (Fuse integrated), (AC-Output->integrated in inverter), USB-Plug+BuckConverter - Dimension:48*23*14mm(L*W*H)
- display (MPPT+BMS) - Dimension:...
-

case06, 2017-10-06:
the layout isnt in /batterycase, but in /boxframe/upkframe ;)

@DhavalGambhava : and btw. the boxframe-freecad files are made with freecad version 0.17, i have it not tested against the 0.16 version. If you have any diffiiculties contact me, or we can meet anyway again online and check things together. Maybe we can try this time the platform Jitsi (https://meet.jit.si/) to ensure that screen-sharing works in both directions ;)

DhavalGambhava, 2017-10-16:
@case06 : Can you please upload UPKframe 8 file again, because when i open this file then some parts are missing!! 
Thxs..

case06, 2017-10-17:
@DhavalGambhava , @mikelovic 

Make sure you use the Freecad version 0.17, then the missing parts should be there .

Concerning the dimensions: Normally it should fit, i think you have modelled the wrong inverter type, cause it has a height of 104 mm, which is typical for the Victron Phoenix 12/800 and the 12/1200 model.

What we use here is probably the 12/350 which has 72mm or the 12/250 which has 84mm height.  I dont know which one Michel has ordered, but in any case we will gain at least 20mm or even 30mmm headspace.  Additionaly we can sink down the top-plate, there is still enough empty space underneath.

I assume it will fit then.  If not, no problem, because i will bring the whole UniProKit with me on Friday, so we can use bigger length for the upward posts .  But i think this will be probably not ecessary.

case06, 2017-10-18:
I have just added the freecad file for the original hammond case which is around the bms and around the mppt .   

https://github.com/CollectiveOpenSourceHardware/LibreSolarBox/blob/master/boxframe/upkframe/hammondcase.fcstd

mikelovic, 2017-11-30:
# ## current state:  @CollectiveOpenSourceHardware/00_reproduction_prototyping 
## modules:
- **electric**: LibreSolar (MPPT+BMS), Inverter (Victron), ESP/Rasbperry
- **Batteries**: LiFePO4 (Calb), 72Ah, 3,2V, 230,5Wh - 4x -> 12,8V, 920Wh
- **Housing/Frame**: UniProKit, with corner elements for blending; wood plates for mounting electric parts and batteries
- **Interface**: Solarinput (MC4), DC-output (car plug), AC-output (schuko, integrated in inverter), main switch

mikelovic, 2017-12-02:
#meeting 2.12.17, fablab st. pauli:
##thinkings:
###interface:
- **solarinput**: problem: plug in/out nicht optimal; welche stecker eignen sich am besten.
-> verschiedene anschlussmöglichkeiten anbieten (vor-/nachteile); optionen: @martinjaeger 
- **USB-Input**: Buck Converter+USB plug, @kruemel  
- **USB-C**: possibilities to integrate in box, @jocogergely
- **RJ45-Jack**: connect more boxes together via CAN protocoll
- **Display**: basic information: SOC, P_in, P_out; 
 - LEDs, 
 - dotmatrix dispay, interface?
 - epaper, interface: SPI, connected to BMS, button to update date (standby Logo)

###electric:
- **inverter:** put outside the box (optional); optimize box layout w/o inverter; if someone needs inverter than connect outside the box

case06, 2017-12-02:
# **Box/housing:**

- ## **UPK**:   add mounting plates for connectors  (already in work).  The outer frame blends can contain cutout-sections, but should not be used for direct mounting of the connectors, mainly for two reasons:

1.  for statical reasons, the give not so much stability as a frame-integrated mounting-plate

2. Handling, maintainance, ergonomic: It should be possible to quickly screw of the blend at one side and then having access to the box-interieur. You don't want to have the blend-plate then still connected by serveral cables with the box.

mikelovic, 2017-12-14:
- [ ]  Battery case @jocogergely 
- [ ]  PCB housing ideas @jocogergely 
- [ ] Solar Plugs
- [ ] USB Buck converter @kruemel 
- [ ] bill of materials

mikelovic, 2017-12-14:
## Bill of material V0.2:
-----------------
###box frame:
- UPK frame:
 - aluminium slot: size defined in 3D model
 - triangle connectors (for stabilization, @case06 whats the right word for it? :))
 - screws, ..., @case06  can you extend the list w/ all needed parts for the Uni Pro Kit w/o corner elements
- wood plates for cover -> thickness like alu profile
- wood plates for internal mounting of parts -> bottum, PCB mounting
- corner elements -> 3D modelled optinons (@jocogergely if you can you can make few suggestions, no prio)
- elect
###electric:
- MPPT, BMS, Gateyway/Server
- buck converter
###interface:
- dc-output plugs (car plug, 2x)
- solar input plugs
- usb plugs
- ON/OFF switch, pushbutton, integrated LED
- RJ45 jacks (2x)
- epaper display+conncetion wire
###connections:
- power wire (6mm^2, red+blue, each 1m)
- balancing wire (2mm^2, 1m)
- communication wire (ethernet cable, 0,5m)
- ferrules, cable lug

closed 2019-03-20 by mikelovic