CollectiveOpenSourceHardware/LibreSolarBox #21: Bill of Materials - V0.2

mikelovic, 2017-12-14:


mikelovic, 2017-12-14:
## Bill of material V0.2:
-----------------
### box frame:
- UPK frame:
  - aluminium slot: size defined in 3D model
  - triangle connectors (for stabilization, @case06 whats the right word for it? :))
  - screws, ..., @case06  can you extend the list w/ all needed parts for the Uni Pro Kit w/o corner elements
- wood plates for cover -> thickness like alu profile
- wood plates for internal mounting of parts -> bottom, PCB mounting, analog V0.1
- corner elements -> 3D modelled options (@jocogergely if you can you can make few suggestions, no prio)
### electric:
- MPPT, BMS, Gateyway/Server
- buck converter
### interface:
- dc-output plugs (car plug, 2x)
- solar input plugs
- usb plugs
- ON/OFF switch, pushbutton, integrated LED
- RJ45 jacks (2x)
- epaper display+conncetion wire
### connections:
- power wire (6mm^2, red+blue, each 1m)
- balancing wire (2mm^2, 1m)
- communication wire (ethernet cable, 0,5m)
- ferrules, cable lug

case06, 2017-12-15:
i checked thingiverse for handles, found the following looking promising:

https://www.thingiverse.com/thing:1228617
https://www.thingiverse.com/thing:130737
https://www.thingiverse.com/thing:12695
https://www.thingiverse.com/thing:2116139
https://www.thingiverse.com/thing:80040

the last one is my favourite.

mikelovic, 2018-01-10:
- [ ] offer from @case06 , UPK, specs from [V6 FreeCAD Design](https://github.com/CollectiveOpenSourceHardware/LibreSolarBox/tree/master/Solar%20Box_FreeCAD%20Design/Solar%20Box%20v_6_boxframe), additionally the wood plates (out-& inside), handles, screws, etc.

- [ ] offer from @martinjaeger , BMS, MPPT, Gateway

- [ ] other electric parts

mikelovic, 2018-01-11:
@case06 :
adaption of box design concerning wood enclosure.
decision for probosal from oliver.
wood plates: bottum and top -> 12mm, side elements -> 5mm, for inner structure

martinjaeger, 2018-02-02:
BOM moved to https://github.com/CollectiveOpenSourceHardware/LibreSolarBox/blob/master/BOM.md

--> Issue closed.

closed 2018-02-02 by martinjaeger