CollectiveOpenSourceHardware/LibreSolarBox #22: Box Design - Industriedesign

mikelovic, 2017-12-21:
in this issue we want to discuss about possible design elements.
**requirements:**
- ecological materials, upcycling elements
- easy production
- moveable
- scalable

case06, 2017-12-22:
some thoughts:

**Ecological materials:**  
We can replace the cover-plates with wood-plates . I would recommend multiplex/plywood of 12mm. Question is how to fix them onto the frame.

**upcycling**

Hmm, in a project like this a more crucial aspect is the overall availability of the components from regular sources and a BOM which list that sources clearly.  But you can maybe upcycle/use any appropriate plate-materials for the cover plates.

**easy production**

Biggest prob is the availability of pcbs in small numbers eg. <= 10 pieces.  Commercial production from watterott is an option starting with 50 pieces, but we need smaller quantities for prototypes and their testing.  

Possible solutions: 
1. Pick&Place-machine   
2. produce smaller batches manually (maybe as "production-workshop"-event at fabulous St.Pauli)

Btw., one thing which is even more important than "easy productions"  is "easy maintainance".

**moveable**

I have added some nice handles to the upklib. See https://wiki.opensourceecology.de/images/8/80/Upk_handles.jpg  and https://wiki.opensourceecology.de/images/6/64/Upk_handle2.jpg

**scalable**

Since the Mosfets are rated at 80A it is theoretically possible to drive it with up to 4 MPPT20s, what means up to 1.6 KWp PV-panels.  In practice we maybe better go with 3  MPPT20s and 60A to stay comfortable under the Mosfets limits equal to 1.2KWp PV.

This means to construct a backplane for the MPPTs, which consists of the CAN-bus data-lines (blue) and two 12V-Power-lines (black and red) like in this diagramm: https://wiki.opensourceecology.de/images/e/e8/Bms12v_systemuebersicht.png

mikelovic, 2018-02-01:
@case06 
can you open the zip file [!here](https://github.com/CollectiveOpenSourceHardware/LibreSolarBox/tree/master/design_proposal) and import it in FreeCAD. Somehow i can't import it correctly in my FreeCAD version (0.16).
thank you in advance

mikelovic, 2018-02-01:
@FWdesign
Feedback:
- der gewichtsschwerpunkt sollte in der mitte liegen. die batterie sollten somit mittig in der box platziert sein.
- optional eine version ohne inverter (blaues modul). damit kann das ganze kleiner produziert werden. -> evtl zwei verschiedene version. oder modular erweiterbar um den inverter. anschluss an die box über DC anschluss.

case06, 2018-02-01:
When i imported the .igs file into freecad 0.17 it became extremely slow ... like swapping.
 Then i changed the view of the whole object into transparent and the performance was normal.

So i saved it into a freecad file 

[https://github.com/CollectiveOpenSourceHardware/LibreSolarBox/blob/master/design_proposal/Solarbox_Trommel_transparent.fcstd](url) 

with transparent view.

see 
[image]

mikelovic, 2018-02-07:
@FWdesign 
weight of batteries. 1.9 kg per cell. So for the 4 cells it is round about 8kg ( exactly 7,6kg).
here you find the specs http://evwest.com/support/CAM72FI%20Final.pdf

mikelovic, 2018-02-08:
@case06 @jocogergely @DhavalGambhava 
hello CAD designer :) we need a new battery holder for this box design. the batteries lie horizontal. so we need again a mounting option and the space holder.
no prio. thank you