CollectiveOpenSourceHardware/LibreSolarBox #26: Battery Case

mikelovic, 2018-01-23:
For the V2 Box we want to adapt the battery mounting inside the box.

mikelovic, 2018-01-23:
[image]

First draft from @case06 [image]

mikelovic, 2018-02-01:
@case06 @jocogergely 
concerning the mounting holes for the battery holder.
we should try to eleminate the wood plate around the batteries. so the vertical mounting hole should be adapted. i tought of mounting it on the aluminium rail (inside so there is space for the blend)

mikelovic, 2018-03-08:
@DhavalGambhava @jocogergely @case06 .
-> so we would need 2 different types of the case
 - for horizontal mounting
 - for vertical mounting

@DhavalGambhava you started with some designs at the last workshop, can you push the files please.

case06, 2018-03-09:
Not sure if i get you right.   To my understanding pure horizontal or pure vertical mounting isnt an option for reasons of stabilty and rigidity, so the case should be anyway mounted in more than one plane. At least on two planes.

Its no big problem to achieve that even when removing the wooden mountplate. I would suggest to indroduce two additional Tslot-Bars within the frame and thats it.  Then you can mount the cellpack anyway in vertical and if wanted as well in horizontal orientation.

DhavalGambhava, 2018-03-11:
@case06 
Box Design till not finished, but I uploaded in GiutHub folder.
later I will Update this design model.

DhavalGambhava, 2018-06-02:
Hi all,
@mikelovic  @case06 @jocogergely 

I just uploaded new design of Battery Case/ Holder. So we don't required wood to hold battery  :-)
@case06, Oliver  As we discuss last time during workshop, so please check the design that it can be printed in 3D printer or not ? 

There is still open point regarding to the **Top cover** of Battery Case, 
@jocogergely  , Jozsef   Is it possible for you to design this part ?

@mikelovic , Michel I design this case according to both design type of box 1) Uni-pro kit Battery box and 2) Round design (by Karlsruhe friends). So, check it out this design, if you have any comments please let me know ;-) 

Thank you and have a nice weekend to all !

case06, 2018-06-04:
@DhavalGambhava   hey, cool stuff, looks great !  And yes, this should be printable, as far as i can see you have taken all critical aspects that we discussed into account.

Maybe there will be a minor imperfection in the holes of the junction-corners but thats probably neglectable (and i would do it the same way) so we can give it a try.

case06, 2018-06-04:
two suggestions for optimizations:

- shaping the junction holes hexagonal for snug-fit with hexagonal nut. I am not sure wether the (round) hole size will be big enough for the nut. But given that, one can make it hot with an soldering iron and melt it perfectly into the hole

- maybe it would be good for stability, to make the "windows" in the wall smaller (and therefore the corner-pillars wider). Reason is the grain-direction: the weak-point in 3d-prints is always the adhesion between layers. So if you have a long structure with a small base-area it can more easily break along the length. A bigger base-area means more layer-adhesion.

But it depends also from other aspects like material-quality and printing-parameters (e.g. temperature), how well the layer-adhesion is.  So we should give it a first try as it is, maybe its already sufficient.

DhavalGambhava, 2018-06-23:
@mikelovic  @case06 
I added Cover in battery case, which protect the connection between batteries. Please look in "Battery Case_Holder_v_4" file.

case06, 2018-06-23:
Looks good, but you should maybe add in the top cover a few holes for the cables :)

DhavalGambhava, 2018-07-04:
@case06   Done!! Thxs for reminder, I added slot for wire in cover.

@mikelovic  @case06  I found it's bit complected to install/ make slot to hold MPPT & BMS on battery case, because battery case wall is like wire frame !! (to reduce material use) & reduired more complex design + to we have to prepare in 3D printer ! 
Please give some suggestion on it, else we can make bar slot and then glue it on Battery case! which is not good :-(

mikelovic, 2018-07-05:
@DhavalGambhava  looks very good! thank you!
Would it be one possibility to mount the electric cases like this?

[image]

case06, 2018-07-05:
A click-option would be very cool, but is maybe not so easy to construct (at least i never did so far). But when thinking about click-option, one could also think about making the two halfes of the Batterycase and its Top-lid as well clickable.  I would see that as a longterm goal/option.

The best way to achieve a good screw-mounting is in my opinion to make the related side wall compact and thicker, about 10 or 12 mm and integrate apprpriate holes, into which the nuts can be embedded.

IOW, we use the Nut as a thread which is implemented in the case and screws can be driven into, without the need to hold/block the nut with a pliers or something.

case06, 2018-07-05:
Another thing: I discussed several weeks ago with Martin, how a future version of the pcb-case should look like, and we came up with a version which is not any longer sandwiched/piggy-backed, but having the pcbs beside each other, for thermal reasons AND we would like to add a heatsink into the case. 

This would imho lead to a conformation like this:  https://wiki.opensourceecology.de/images/4/4e/Photo_2018-07-05_14-17-40.jpg

 And btw. the ffc foil-cable and its connector should then be replaced with a regular one, because its more expensive, difficult to order and can break more easily, so there is no real reason or advantage in a  ffc-cable.

closed 2018-08-25 by mikelovic