CollectiveOpenSourceHardware/LibreSolarBox #27: Testing - Libre Solar Box

mikelovic, 2018-02-01:


mikelovic, 2018-02-01:
specification of used solar panels at the fablab.
see: https://www.jws-store.de/solarenergie/solarpanels-mono-12volt/55watt-solarpanel-12-volt-monokristallin/a-194/

## Technische Daten:
Nennleistung Pmax 140 Watt
Spannung bei Nennleistung Vpmax 22 Volt
Leerlauf Spannung Voc 26 Volt
Kurzschluss Strom Isc 6,87 Ampere
Strom bei Nennleistung Ipmax 6,36 Ampere
Temperaturbereich -40°C / + 85°C
Toleranz +/-3 %
Flexible monokristalline hochleistungs Solarzellen von SUNPOWER, Wirkungsgrad über 21%
Abmessungen: 1490 x 560 x 3mm
Gewicht 2 kg

mikelovic, 2018-04-05:
## Meeting 05.04

*#1* Testing of components (MPPT, BMS)
- what to test, how to test, when to test

*#2* Installation of Energy System in Fablab St.Pauli
- which loads, how much solar power, how much capacity -> design and layout

*#3* Monitoring
- energy data monitoring and analysis

mikelovic, 2018-04-12:
## Meeting 12.04

basic concept
[image]

concept for discussion @CollectiveOpenSourceHardware/00_reproduction_prototyping

SchneiderAn, 2018-04-19:
[Testing MPPT.pdf](https://github.com/CollectiveOpenSourceHardware/LibreSolarBox/files/1928263/Testing.MPPT.pdf)

SchneiderAn, 2018-04-19:
[image]

SchneiderAn, 2018-04-19:
Testing the BMS: 
1.	find out the BMS requirements / What does the BMS needs to do:
•	Voltage balancing 
•	Temperature checking
•	etc
2.	Testing if the BMS fulfils each requirement:
•	Measure the voltage at each cell 
etc

mikelovic, 2018-05-17:
Messplan:

### MPPT
- min/max Strom: 0..20A
- min/max Spannung: 16V..55V
- Effizienzpunkte (eta(P))
- Spannungsqualität am Ausgang (Regelverhalten bei Sprung am Eingang)
- MPPT Qualität (Abweichung, Zeit)

### BMS
- min/max Strom: 0..30A
- (Schutzmechanismen - Ladeschlussspannung, Tiefentladeschutz)
- Effizienzpunkte (eta(P))